sparkfun/SparkFun_AS7265x_Arduino_Library #1: Reading calibrated data always selects the same device

TheNimmer, 2018-11-27:
The calibrated results functions all duplicate the first devices data. 

In other words, `getCalibratedA` returns the same value as `getCalibratedG` because there is no call to `selectDevice` in the `getCalibratedValue` function.

I'm assuming `getCalibratedValue` should be more like the `getChannel` where it accepts the proper device as an argument.

nseidle, 2018-11-28:
Yep! Good catch. getCalibratedValue() now calls appropriate device. I've rolled the library version. Please install v1.0.1 and let me know if you have any more issues.

closed 2018-11-28 by nseidle

Michael-Raven, 2018-12-03:
The effect of this bug is very evident in the sample data, and graphs, in the Hookup Guide