sparkfun/SparkFun_AS7265x_Arduino_Library #3: NIR data returned for VISIBLE bank & vice versa

Michael-Raven, 2018-12-03:
#define AS72651_NIR     0x00
#define AS72652_VISIBLE     0x01
#define AS72653_UV      0x02

for context, here are the (correct) bulb device assignments

#define AS7265x_LED_WHITE	0x00 //White LED is connected to x51
#define AS7265x_LED_IR	0x01 //IR LED is connected to x52
#define AS7265x_LED_UV	0x02 //UV LED is connected to x53

nseidle, 2018-12-03:
I'm unclear as to what your issue is. You are calling what function and expecting what values?

You are probably already aware, but the x51 (channels 610, 680, 730, 760, 810, 860um) and x52 (channels 560, 585, 645, 705, 900, 940) overlap in their spectral channels so it's best to take a full 18 channel reading rather than reading a specific sensor.

The white LED just happens to hang off of the x51 IC. It could have been attached to any of the three. Same goes for the IR LED.

Michael-Raven, 2018-12-04:
Thanks, Nate; yes, I'm quite clear on all of that.

If I call 'GetCalibratedR', for example, I expect to get back 610 nm
data from the x51 (see data sheet, figs. 15 & 16). The newly-repaired
library code reads:

float AS7265X::getCalibratedR() {
  return (getCalibratedValue(AS7265X_R_G_A_CAL, AS72651_NIR));
}

And since 'AS72651_NIR' equates to a device number of 0x00, that is
indeed what I get. No bug. My complaint is about the confusing
nomenclature: device 0x00 is the AS72651, the visible light member of
the triad. 'AS72651_NIR' is a contradiction in terms. Same thing goes
for 'AS72652_VISIBLE'.

Perhaps not worth changing, since the answers are coming out correctly.
Or perhaps in your mind the x51 really *is* the NIR member of the triad,
and the x52 centers on the visible (overlapping spectra, as you say).
But that's not how the datasheet is written---or color-coded. In any
case, at least one reader of your code (me) was bogged down for five or
ten minutes, distracted by the need to cross-check #define's, etc., to
be sure everything works as advertised.

nseidle, 2018-12-04:
Ahh! I see now. You are treading dangerously into PR territory ;) If you can see a better way to make the code more readable, I'm all for it. But I agree, the current define setup is fairly arbitrary.

Michael-Raven, 2018-12-04:
My suggestion, of course, is simply to: change 'AS72651_NIR' to 'AS72651_VISIBLE' and 'AS72652_VISIBLE' to 'AS72652_NIR' in the header file---or perhaps dispense with the second half of each name altogether (call them all ..._ADDR)---and then make corresponding changes to the code.  Two minutes work, if you decide it's worth it.

The benefit, as I see it, is that your readers would then be able to read the code straight through, going through that mental checklist you always maintain: "yep, that looks right...", instead of having to stop and flip back & forth between '.h' & '.cpp' upon encountering the oxymoronic name choices. The latter place (admittedly minor) demands on the reader's limited memory stack, and thus impede understanding.

Anyway, enough preaching. I'm perfectly happy to let you decide (it's your code, after all), and close this issue out.  And finally, let me say I'm having loads of fun with this product! Thanks for making the chipset so easy to use.

nseidle, 2018-12-06:
Thanks for the feedback! I'm going to close it out but if anyone else gets confused I'll rename.

closed 2018-12-06 by nseidle