sparkfun/SparkFun_AS7265x_Arduino_Library #4: isConnected problem

JRDavisUF, 2019-03-05:
Using my Win10, CCSv8, Msp432,  and Energia, I need to add a write statement after the I2C begin in order to get the device to respond.  I've (and many others) have run into this issue before with I2C...although I forget the reason offhand why this is needed.

... 
 _i2cPort->beginTransmission((uint8_t)AS7265X_ADDR);
  _i2cPort->write(0x00); // <===== Needs this
  if (_i2cPort->endTransmission() != 0)
....

nseidle, 2019-03-05:
Sounds like a weird bug with MSP432 I2C implementation. 

If you have a preprocessor flag that we can test against, we can add this to the library. [Here's](https://github.com/sparkfun/SparkFun_BNO080_Arduino_Library/blob/master/src/SparkFun_BNO080_Arduino_Library.h#L46) a rough example of what I'm talking about.

PRs are always appreciated but this is not too complicated.

JRDavisUF, 2019-03-05:
I think it's more of a weird Energia problem as opposed to a MSP432-specific one, for example, see:  https://github.com/energia/Energia/issues/387   https://github.com/energia/Energia/issues/617   

I believe the added write works fine with non-Energia system, but I don't have anything to test.  But if a Energia-specific solution is desired, then something like this works to fix the issue:
...
  _i2cPort->beginTransmission((uint8_t)AS7265X_ADDR);
#ifdef  ENERGIA             //  New
  _i2cPort->write(0x00);  //  New
#endif                            //  New
  if (_i2cPort->endTransmission() != 0)
...