sparkfun/SparkFun_AS7265x_Arduino_Library #5: AS7265x datasheet and Spectral Triad schematic do not agree - which is accurate?

Daemach, 2019-04-05:
This isn't the best place for this question, but the Sparkfun forums don't seem to be monitored by the office and I'm really hoping for a good answer.  I'm interested in the spectral triad, but the pinout in the datasheet does not match the schematic.  Based on the datasheet, the triad board should not work.  At all.  One prime example is that the inter-chip i2c pins on the as72651 are listed as 9&10 in the [datasheet](https://cdn.sparkfun.com/assets/c/2/9/0/a/AS7265x_Datasheet.pdf) but the [schematic ](https://cdn.sparkfun.com/assets/3/b/0/3/e/Qwiic_Spectral_Sensor-AS7265x.pdf) uses pins 20 and 1.  There are a number of similar discrepancies, but this alone is enough to make it non-functional.  I noticed there are a couple of posts from users in February, that have 0 replies, btw, that couldn't get this board working.

I could also use information on how to update the firmware on the SPI flash - there is a programming header, but no mention of how this gets hooked up, and a note in the schematic that says the AS7261 (I assume that's a typo) must be kept in reset while programming, but there is no provision to do that.  RST and INT are tied together with a jumper, but I don't understand why that is, either.  And the AS72651 has an internal pull-up on reset according to the datasheet so the 10K external pull-up doesn't make sense to me.  I'm sure that the error must be mine - Nathan Seidle is famous for knowing his stuff.  I sure could use some guidance and clarification!

nseidle, 2019-04-05:
AMS has not corrected their datasheet for this product. Check their [design considerations](https://ams.com/documents/20143/36005/AS7265x_UG000337_2-00.pdf/7af9f359-e2a4-f286-0e12-6c9e4e89c054) doc; it has the pins correctly shown. We laid out our first rev following their "[v1-04] 2018-Jul-09" datasheet, only to find out they switched the pinout in production. 

Reprogramming the SPI flash can be accomplished my using our [ISP kit](https://www.sparkfun.com/products/11591) but modifying it for SPI flash EEPROMs (rather than AVRs). You'll need a EEPROM programmer but these are *really* common and low-cost. Our programming jig uses the exposed RST and GND pins located on the edge 0.1" header to hold the AS72651 in reset.

closed 2019-04-05 by nseidle

Daemach, 2019-04-05:
Thank you.  What is the purpose of the jumper between INT and RST?'

Will this work directly with both the firmware and your ISP kit?  https://www.embeddedcomputers.net/products/FlashcatUSB/  Any changes necessary?

nseidle, 2019-04-05:
[image]

I don't see a jumper between INT and RST. That would be pretty weird. What jumper are you speaking of?

Flashcat looks like it would work. You'll just need to wire MISO to the correct pin that touches the MISO pad on the landing pads on the PCB. I recommend opening up the Eagle files so you can be sure you're hitting the correct holes. Also, I'm pretty sure the board is 3.3V so be sure not to expose the board to 5V (I'm not sure what voltage the Flashcat runs at).

Daemach, 2019-04-05:
I thought it was weird too - this one:  https://www.screencast.com/t/2hVmbC5XonD7

Can you recommend a programmer you trust at the same or better price point?

Daemach, 2019-04-05:
Maybe it isn't a jumper but any time I see 2 pins together that's what I think of.

nseidle, 2019-04-05:
Ah - that's just the eagle 2 pin connector.

I bought my CH341 based programmer off Amazon for initial programming and testing. This software is *really* bad tho. For production we use the ATmega on the test jig to program the SPI EEPROM over, well, SPI.

Daemach, 2019-04-05:
Gotcha - that makes more sense.  Really nice design on that board, btw.  Next-level Eagle skills.

Final question, I hope - it looks like you used the gen 1 pinout from the design considerations doc for the triad board.  Are there any drawbacks with gen 1 vs gen 2?  Does the library work with both generations of firmware?  If you were going to build a new design with these sensors would you go with gen 1 or gen 2?

nseidle, 2019-04-08:
Thanks. I don't recommend making circular layouts. Eagle is not very suited for it.

I'm not sure what you mean by gen 1 vs gen 2. Follow the pinout in our schematic and use AMS's latest firmware otherwise you run the risk of having all sorts of problems.

ldab, 2019-07-30:
Should we revisit this as the [errata](https://ams.com/documents/20143/36005/AS7265x_ES000127_1-00.pdf/49aff479-3b43-f541-2d52-02a7f7d0c3e1) mention some Pin names misplaced?