sparkfun/SparkFun_AS7265x_Arduino_Library #14: Stuck in while(1) loop

Haismd03, 2022-04-15:
I have experienced this issue many time. It happens randomly within 30 sec to 5 hours and it completely blocks around program.

I have located, that the problem is in library method called `virtualReadRegister()` (possibly in `virtualWriteRegister()` too). In this method program gets stuck in `while(1)` loop until some bit is cleared. Problem is, that there is certain chance, that this bit won't be changed and program will be stuck in this method forever.

For now, without modification this library is pretty much unusable, due to this problem.

(I've been working on Raspberry Pi Pico, using core from earlephilhower [(GitHub)](https://github.com/earlephilhower/arduino-pico) programmed on Arduino IDE 2.0)

nseidle, 2022-04-15:
Agreed. I've added timeouts. The sensor will report 0 if a device fails to respond within 150% of the integration time (210ms by default). Please give the [latest commit](https://github.com/sparkfun/SparkFun_AS7265x_Arduino_Library/commit/95b7893137312c7ff90eebbad1e394364303aeb5) a test and let me know if it fixes the problem.

Haismd03, 2022-04-16:
I've tried newly committed library and my program still gets stuck. I need to do more debugging to see, why is this happening. Unfortunately, I won't be able to do so probably until Sunday.

Sorry
I'm sure we'll find reason for it :wink:. Thanks for your help

nseidle, 2022-04-16:
Sound good. Please report back so we know whether to release the fix or leave it out.

rabzgg, 2022-10-21:
are the measurements suppose to be printed instantly? because mine only printed "AS7265x Spectral Triad Example"

cool-benson, 2023-08-09:
@nseidle I found that on line 591(`startTime = 0;`)  seems to be wrong, instead it should be `startTime = millis();`. LMK what you think, I can create a PR if needed.

ayavilevich, 2023-09-08:
I can confirm the bug and the solution.

nseidle, 2023-11-29:
Thank you! Yup, the startTime should not be set to 0. Please give the latest release a test. Closing for now. Please open a new issue if the problem persists.