sparkfun/SparkFun_AS7265x_Arduino_Library #15: What is the basis of the calibration value?

harlan56, 2022-08-23:
Calibrated values are returned from getCalibratedValue*(), what basis or conditions are the calibration values based on?

cmosthings, 2023-11-15:
I agree having some Detail as tho what the Calibration even accounts for would be nice to have. Especially wheather it accounts for temperature changes or not.

cmosthings, 2023-11-22:
I took a look at the Serial interface for the chip what i found is:
- you can Read the Calibration Values from the Chip with ATSCLx (x=0..17)
- the relationship between Raw and Calibrated is just  RAW*Calibration = Calibrated. 
- the Calibration Values don't change with Temperature
For Reference Here are the Cal Values i got from my Sensor= 1.1586,1.0509,0.7920,0.8261,0.8532,1.1863,0.5587,0.5182,0.4461,0.4562,0.6088,0.9338,0.7763,0.9399,0.9088,0.8310,0.7110,0.6935

nseidle, 2023-11-29:
Please see [the datasheet](https://ams.com/documents/20143/36005/AS7265x_DS000612_1-00.pdf/08051c8a-a7f6-6231-7993-2d3fe0bf38b8) (page labeled 29) for more information about the calibrated values. These are set by the manufacturer.

closed 2023-11-29 by nseidle