sparkfun/SparkFun_AS7265x_Arduino_Library #17: Not getting any readings

Ajaynithish, 2022-12-13:
I have the spectral triad to my Wemos lolin32 board through i2c and I have programmed the ReadRawValues program. All I get is a bunch of zeros in the output. Tried with every other example program too still got only zeros. Changed the controller to arduino nano and sparkfunfun ESP8266 thing dev board and still getting only zeros. 

Where is the issue?

candyman2903, 2023-04-13:
Same mistake here.

Bohai-Ruan, 2023-08-02:
I had luck by downgrading to 1.0.2 .

ayavilevich, 2023-09-08:
This is caused by a bug in the timeout check.
See https://github.com/sparkfun/SparkFun_AS7265x_Arduino_Library/issues/14

nseidle, 2023-11-29:
Thank you! Yup, the startTime should not be set to 0. Please give the latest release a test. Closing for now. Please open a new issue if the problem persists.

closed 2023-11-29 by nseidle