sparkfun/SparkFun_AS7265x_Arduino_Library #18: AS7265X::get'X'

cmosthings, 2023-11-15:


cmosthings, 2023-11-22:
To answear my own question Serial is factor 2X faster. Still nowhere near the theoretically possible 180Hz.
I tuned the Speed over I2C to about 10Hz with Raw Values and Over Serial I got 20Hz just out of the box.

The key to get the I2C to be faster is tho reduce the delay(AS7265X_POLLING_DELAY); to a shorter Period 1ms should be already better. I reduced it to 100millis for myself instead of the 5ms and it worked great.

nseidle, 2023-11-29:
Did you adjust the integration time? By default, the integration time is 0x20 or 32*1.2.78ms = 88.96ms. This matches your 10Hz I2C experience. Consider decreasing setIntegrationCycles() to achieve the integration time you desire. 

[image]

Datasheet page [label 32](https://ams.com/documents/20143/36005/AS7265x_DS000612_1-00.pdf/08051c8a-a7f6-6231-7993-2d3fe0bf38b8).

closed 2023-11-29 by nseidle

cmosthings, 2023-12-28:
Yes I did setIntegrationCycles(1) and the Hz is still stuck at what I mentioned before. I Measured the time between resetting the MeasurementMode(AS7265X_MEASUREMENT_MODE_6CHAN_ONE_SHOT) and dataAvailable to approximately 2*2.78ms which is great and as expected. Getting the data from the Sensor over I2C seems to be the major Bottleneck limiting to cycle times of over 100ms this is the issue I and others are having.