sparkfun/SparkFun_AS7265x_Arduino_Library #19: Sensor not detected with basic I2C wiring - time window issue?

vitara-tran, 2024-01-18:
# Issue Description

I am having an issue connecting to the AS7265x sensor in time, when i use the  [Example1_BasicReadings.ino](https://github.com/sparkfun/SparkFun_AS7265x_Arduino_Library/blob/main/examples/Example1_BasicReadings/Example1_BasicReadings.ino) program it never sees the sensor even after wiring correctly. 

I could only get it to detect the sensor by putting the connection command in a while loop and then physically pulling the power cable to the sensor in and out until it worked. 
Here's my connection code:
```
  Serial.begin(115200);
  Serial.println("AS7265x Spectral Triad Example");
  bool loop_run = true;
  while (loop_run) {
    if (sensor.begin() == false)
    {
      Serial.println("Connection Failure");
    } else {
      Serial.println("Connection Success");
      loop_run = false;
    }
  }
```

I think there is some sort of connection time window I'm missing, can you please let me know if there's anything additional I have to add to the code to make it connect reliably and reproducibly. 

# Steps to Reproduce
Arduino: Due
- Wire SDA -> pin20, SCL -> pin21, 3v3 -> 3v3, GND -> GND
- Upload the Example1 BasicReadings sketch
- Observe that sensor is not picked up

nseidle, 2024-01-18:
There is not a time window. The device will acknowledge over I2C when it is present on the bus and powered.

Pulling/unplugging the cable will cause the I2C interface to act as if the IC has acknowledged, when it has not. You're getting false positives.

Please check your hardware. There may be a problem with the board, connectors, etc. I would use a multimeter to verify continuity from the connections on the board (3.3V, GND, SDA, and SCL) back to the microcontroller.

closed 2024-01-18 by nseidle

cmosthings, 2024-01-18:
Apparently there is an Issue with the DUE and the Wire.h lib that's just best avoided by switching to other Hardware. Some People seem to be successful with using a different I2C library ¯\_(ツ)_/¯.

vitara-slipenchuk, 2024-01-19:
> There is not a time window. The device will acknowledge over I2C when it is present on the bus and powered.
>
> Pulling/unplugging the cable will cause the I2C interface to act as if the IC has [...]

@nseidle Hey thanks for your response. We ran this on a fresh AS7265X and verified continuity along all pins, thanks for suggesting that.

We found that the sensor was failing because maxWaitTime is initialized as 0, and then when begin() is run it calls virtualReadRegister(), which then evaluates:
```
//Wait for WRITE flag to clear
  unsigned long startTime = millis();
  while (1)
  {
    if(millis() - startTime > maxWaitTime) return(0); //Sensor failed to respond
    status = readRegister(AS7265X_STATUS_REG);
    if ((status & AS7265X_TX_VALID) == 0)
      break; // If TX bit is clear, it is ok to write
    delay(AS7265X_POLLING_DELAY);
  }
```

As you can see ```millis() - startTime > maxWaitTime``` will always evaluate to true because maxWaitTime is set at 0 at this point, and unless time starts flowing backwards, the code will always break here.

I believe this bug was introduced in #14 where you took the suggestion of cool-benson which fixed the bug that would make ```millis() - startTime > maxWaitTime``` always evaluate to false. 

I went ahead and modified the header file to init maxWaitTime with value 1000 (1 second) and the example is now working.

We tested this on an Uno as well.