sparkfun/SparkFun_AS7265x_Arduino_Library #20: AS7265x Spectral Triad Example Sensor does not appear to be connected

obisboo, 2024-03-02:
Hello,

"AS7265x Spectral Triad Example Sensor does not appear to be connected. Please check wiring. Freezing..."

This is the message when I tried to use the sensor, also I tired with your example codes, to exclude the problem from my code. Sometimes works, but after a reset I received again the freezing message.  It was working randomly, I just push reset on Arduino until the sensor start to read, but in most of the cases was not working. 

After a long debugging, replace the Arduino, sensors, wires, I found the problem was with the version of the lib, in my project with v1.04 I have the freeze issues, but with v1.03 is working perfectly. 

I just want to tell you what was the problem in my side, maybe others have the same issue, and for the moment this was my solution I could find.

Thank you for your work guys.
Have a wonderful day.

Asadsaeed003091, 2024-09-20:
Thanks, i had the same issue for days, unluckly i changed the libray but i only tried 1.0.0 and 1.0.4 and had same issue with both of them, so i though the library is not causing that issue. But with 1.0.3 everything is working fine.

2ck, 2024-10-10:
Same problem here, looks like commit 95b7893 broke things.
It introduces a timeout, but the default value of 0 is already used before it is set with `setIntegrationCycles`, leading to an early abort after 1ms with no data available

```
  unsigned long startTime = millis();
  while (dataAvailable() == false)
  {
    if(millis() - startTime > maxWaitTime) return; //Sensor failed to respond
```

aryz-ssh, 2024-10-23:
> Thanks, i had the same issue for days, unluckly i changed the libray but i only tried 1.0.0 and 1.0.4 and had same issue with both of them, so i though the library is not causing that issue. But with 1.0.3 everything is working fine.

Had a similar problem as obisboo. Switched library from 1.0.3 to 1.0.4 and now works perfectly fine (for now at least). Thanks for the suggestion!

scottkildall, 2024-11-21:
confirmed, same problem. 1.0.3 fixes it.

nseidle, 2024-12-11:
Sorry for my breaking change. Should be fixed in v1.0.5 now, thanks to @malcolm-sparkfun.

closed 2024-12-11 by nseidle