sparkfun/SparkFun_AS7265x_Arduino_Library #21: Strange behavior on results of common lamps and comparing with Sekonic

kiromaisus, 2024-05-29:
Hi.
I've been trying to make the sensor work, tried 2 libraries, many tests, comparison with Sekonic C-800, and this is the spectrum I get when an incandescent lamp shines on the diffuser (I previously checked that it just attenuates a bit the same way in every wavelength) and then to the sensor, which is close.

[image]

I also tried with other lamps, and they work better, it's only when it should be very little amplitude that there is some more, but it could be light from the side of the box.

 I also uploaded the same firmware (of the sensor) to it, but nothing really changed. I'm sure it uploaded because the program read the data that was sent as well. I'm waiting for Osram to respond with the firmwares to try once again.

This is the LED lamp I also test on. Just the channel 9 is very far away from the point it should be.
(Orange is what the theory said using Sekonic C-800, and blue is what I measured).
[image]

Does anyone know what could be? If I try to manually calibrate and put gains outside of the library, it's impossible to obtain a reasonable calibration, because one channel requires gain for one lamp, and attenuation for the other lamp, few channels require both attenuation or gain (which is doable).

Thank you in advance for your attention.

cmosthings, 2024-05-29:
Hi, 
Could you please provide the code you use to obtain these values?
What happens to your value 9 is hard to say could be noise could be a led somewhere. 
The Sensor has 18 channels total you show 14 could you label channels with what nm they are? What kind of diffuser are you using? Are you using a window of some sort? Is there a reflecting material nearby that happens to reflect channel 9 type light?

kiromaisus, 2024-05-29:
The part in the code that interferes with the sensor is this:

On the setup:
```
  // Once the sensor is started we can increase the I2C speed
  Wire.setClock(400000);

  // There are four gain settings. It is possible to saturate the reading so don't simply jump to 64x.
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  // sensor.setGain(AS7265X_GAIN_1X); // Default
  // sensor.setGain(AS7265X_GAIN_37X); //This is 3.7x
  // sensor.setGain(AS7265X_GAIN_16X);
  sensor.setGain(AS7265X_GAIN_64X);
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

  // There are four measurement modes - the datasheet describes it best
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  // sensor.setMeasurementMode(AS7265X_MEASUREMENT_MODE_4CHAN); //Channels STUV on x51
  // sensor.setMeasurementMode(AS7265X_MEASUREMENT_MODE_4CHAN_2); //Channels RTUW on x51
  // sensor.setMeasurementMode(AS7265X_MEASUREMENT_MODE_6CHAN_CONTINUOUS); //All 6 channels on all devices
  sensor.setMeasurementMode(AS7265X_MEASUREMENT_MODE_6CHAN_ONE_SHOT); // Default: All 6 channels, all devices, just once
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

  // Integration cycles is from 0 (2.78ms) to 255 (711ms)
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  sensor.setIntegrationCycles(49); // Default 50*2.8ms = 140ms per reading
  // sensor.setIntegrationCycles(1); // 2*2.8ms = 5.6ms per reading
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

  // Drive current can be set for each LED
  // 4 levels: 12.5, 25, 50, and 100mA
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  // White LED has max forward current of 120mA
  sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_12_5MA, AS7265x_LED_WHITE); // Default
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_25MA, AS7265x_LED_WHITE); //Allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_50MA, AS7265x_LED_WHITE); //Allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_100MA, AS7265x_LED_WHITE); //Allowed

  // UV LED has max forward current of 30mA so do not set the drive current higher
  sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_12_5MA, AS7265x_LED_UV); // Default
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_25MA, AS7265x_LED_UV-bad); //Not allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_50MA, AS7265x_LED_UV-bad); //Not allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_100MA, AS7265x_LED_UV-bad); //Not allowed

  // IR LED has max forward current of 65mA
  sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_12_5MA, AS7265x_LED_IR); // Default
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_25MA, AS7265x_LED_IR); //Allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_50MA, AS7265x_LED_IR); //Allowed
  // sensor.setBulbCurrent(AS7265X_LED_CURRENT_LIMIT_100MA, AS7265x_LED_IR-bad); //Not allowed
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

  // The status indicator (Blue LED) can be enabled/disabled and have its current set
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  // sensor.enableIndicator(); // Default
  sensor.disableIndicator();

  sensor.setIndicatorCurrent(AS7265X_INDICATOR_CURRENT_LIMIT_1MA);
  // sensor.setIndicatorCurrent(AS7265X_INDICATOR_CURRENT_LIMIT_2MA);
  // sensor.setIndicatorCurrent(AS7265X_INDICATOR_CURRENT_LIMIT_4MA);
  // sensor.setIndicatorCurrent(AS7265X_INDICATOR_CURRENT_LIMIT_8MA); // Default
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-

  // The interrupt pin is active low and can be enabled or disabled
  //-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-=-
  // sensor.enableInterrupt(); // Default
  sensor.disableInterrupt();
```

On the loop:
```
[... 20 lines cut ...]
  canal_espectrometro[16] = sensor.getCalibratedK();
  canal_espectrometro[17] = sensor.getCalibratedL();
  printf("\n\n");
  for (int i = 0; i < 14; i++)
  {
    printf("%f\n", canal_espectrometro[i]);
  }
delay(3000);
```

I forgot to show the other 4 channels, but they are infrared and doesn't matter to my use.

The 14 channels were ordered in ascending order of wavelength, as the code shows now.

I'm using a diffuser I bought on internet for lightning, it's about 1 mm thick and covers the 3 sensors with ease. It has a milky appearance and it's not very light blocking.

I'm using a box with the lights inside on top, they shine down on a part where I put the sensor. There is only a small error (when I turn off the lamps) from the ambient light, which is few counts, less than 3 counts (only penumbra from the ambient). And for when the lamps are on, I check with the Sekonic and it seems no different than what it should be, it's the lamp spectrum in any position.

I verified the diffuser measuring the incandescent light with the Sekonic in the same spot with and without the diffuser, and ran this app https://apps.automeris.io/wpd/ to get the data and saw it was nearly only one value of attenuation for all frequencies. And even if it was the diffuser, it's the same for both lights, and I take care to don't touch it when measuring the 2 lights (one each time) so it won't change position. If it was something with the diffuser, it would appear for both lamps, which can be only for channel 9.

Not using any window.

There isn't any material inside that could be reflecting off the light, it's a wood box, it probably has some particular spectrum, but Sekonic shows only the almost perfect incandescent spectrum with temperature of about 2450 K, and CRI of about 98.X.

I can show you the data I collected (that I didn't show yet) if you want, as how I theorized how much each channel should read, based on the gaussian curve of each channel, the data I got from Sekonic, and the calculation (integration) of the spectrum for each channel.